Ultimaker/Cura #14166: Tree supports create islands in mid air.

spiritparticle, 2022-12-30:
### Application Version

5.3.0-alpha-xmas

### Platform

Windows 10

### Printer

Creality CR-10

### Reproduction steps

Leaving all tree settings at default.
Slicing object using tree supports.
As I explore the layers I see many instances of "islands" in mid air. Probably as a result of the tree algorithm.

### Actual results

Here is a [screen capture](https://imgur.com/TWFH90r) of the islands in the layers.
Here is [another screen capture](https://imgur.com/undefined) of the unusual situation of the ends of the trees missing the lowest part of an overhang and generating one large support roof in a position that would cause the project to fail printing also. The ends of the branches seem to favour the sides of the overhang not the bottom.

### Expected results

I would expect the tree creation process would implement something to avoid these islands, delete them. Or the process would ensure concentric layers to produce branching. It looks to me like it might be some kind of lofting process using circle packing or something.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Dripper6.zip](https://github.com/Ultimaker/Cura/files/10322687/Dripper6.zip)

GregValiant, 2023-01-03:
Thanks for the report.
_"Leaving all tree settings at default."_
Every model is going to be at least a little different.  There are 73 support settings and some will need to be adjusted to get the supports the way you need them.  "Islands" forming in the middle of a support structure can be caused by the combination of certain settings like  Support Infill Density and the Tree Support Branch Angle (to name two) as well as the geometry of the specific model.  If you were to load a different model then the "default" settings might work perfectly.

This doesn't sound like a bug in Cura, but rather a settings issue.  If you alter some settings I'm sure you can get the supports to generate correctly so no islands develop.  There is no combination of settings that would be "Universal" and work on every possible model out there.  The "Default" settings are "One Size Fits Many" and not "One Size Fits All".

I'll remove the bug label and change this to a discussion.  If you think that is premature or can point to a setting and say that it doesn't work then this can certainly be revisited.  I'll leave this open so you can respond.

ThomasRahm, 2023-01-03:
See this bug report: https://github.com/ThomasRahm/CuraEngine/issues/9
It illustrates this issue better.

GregValiant, 2023-01-04:
Thanks for the link.
I'll put the bug label back on this for you.  The Cura team may view this differently but my view is that there currently are settings that will mitigate what is going on and keep the floaters from developing.  The density of any tree support structure will increase as it approaches the roof and there must be a sufficient mass of material below to hold up that dense upper portion and any support interface.