Ultimaker/Cura #14170: moving a files crashes cura, for months, not before

Halrardo, 2022-12-30:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 5

### Reproduction steps

Open up cura, its fine i can navigate around and select different menus without crashing.
I upload any stls its fine.
When I go to move stls, it does not respond and wants to white screen.

This has been fine last year with multiple versions. for the last few releases, this problem happens.

### Actual results

When I go to move stls, it does not respond and wants to white screen.

### Expected results

Cura working would be nice. I should be able to move stls.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-03:
Thanks for the report.
I think the Cura Team will need more information about your system.  Are you using multiple monitors?  If you are - are they all of the same manufacturer?  Is this a desktop computer or a laptop?  What video system is used?

Halrardo, 2023-01-03:
Hi, thanks for responding, I belive the video system is NVIDIA GeForce GTX
1050 graphics. I am using a laptop called ERAZER gaming notebook (MEDION),
I upgraded it to 16GB RAM. I'll repeat that last year Cura was working fine
and that these last few months I have out off contacting you about it as I
hoped updates would fix it. I have since found that opening 2 cura windows
helps it for reasons unknown to me.

GregValiant, 2023-01-04:
_"I have since found that opening 2 cura windows helps"_
That's the sort of information that can help chase a problem down.  The Cura team will take a look.

MariMakes, 2023-01-05:
Hey @Halrardo,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for the extra information. 

I think it might be related to https://github.com/Ultimaker/Cura/issues/13820

To learn better about what's slowing your software down, could you install a plug-in that performs more extensive logging? 

You can download it here: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging
After it's installed can you.... 
	1. Select Extensions
	2. Select Remote Support
	3. Start CPU profiler
	4. Try to move files
	5. Stop CPU profiler
	6. Select Save Remote Support Archive
	7. Go to the Marketplace
	8. Select the gear icon 
	9. Uninstall Startup Optimiser
	10. Share the Remote Support Archive Zip file here on Github

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-01-19 by github-actions[bot]