Ultimaker/Cura #14171: Spelling error

santanig5, 2022-12-30:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality CR-10S Pro

### Reproduction steps

Noticed a spelling error in the notification box

[image]

### Actual results

N/A

### Expected results

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

MariMakes, 2023-01-05:
Hey @santanig5,

Thanks for your report 👍

I believe what you are seeing is [a Kerning issue.](https://en.wikipedia.org/wiki/Kerning)
It's supposed to say clear but the c and l are pushed together.

We've seen some more reports about this but often it only shows up for a short time and disappears the next time you check it out. 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

MariMakes, 2023-01-10:
Hey @santanig5,

I brought your issue up to the team. 
We believe we will not work on this anytime soon and will therefore close this issue.

I hope you understand. Thanks for the report. 👍

closed 2023-01-10 by MariMakes