Ultimaker/Cura #14174: Save to disk gcode does not put any file extention on for linux

protocolkey, 2022-12-30:
### Application Version

5.2.1

### Platform

Mint Linux cat 20.3

### Printer

Longer LK5 PRO

### Reproduction steps

Drag and drop STL onto platform, slice in generic PLA Standard Quality. save to linux disk with EXT4 partition.  older versions automatically added file extension.
screen shot of saving
[image]

Screenshot of file saved and a manual copy of the file and naming showing proper file extention
[image]

### Actual results

Gcode file saved without extention, Gcode file seems fine and works as expected if extension is added.

### Expected results

Expect selected file type in dialog drop down to check and append proper extension if it does not already have it in the manual text box entry.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]

MariMakes, 2023-01-05:
Hey @protocolkey,

Welcome to the Ultimaker Cura Github 🚀 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12659
We've had multiple reports on this issue, we have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12659

Fingers crossed that we found a solution soon. 🤞

closed 2023-01-05 by MariMakes

MariMakes, 2023-03-31:
Quick update on our side. 

We might have a fix for this issue. 🎉 
It's a bit hard for us to test, but we have a nightly build available. 
Can you please try it out and let us know if it resolves your issue? 

[You can find the download links and the discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)