Ultimaker/Cura #14176: Constantly Crashing

B18rexracer, 2022-12-31:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 S1 Pro

### Reproduction steps

Cura crashes constantly, sometimes while just opening.  I changed display setting as advised.  Now it crashes less but still 7/10.  It crashes at various times- sometimes when clicking "Preview" sometimes when trying to open the print setting pane.  I've rebooted, I've reinstalled the program and nothing seems to help.
[cura.log](https://github.com/Ultimaker/Cura/files/10327094/cura.log)

### Actual results

Initial changes with the display in windows settings made it so it I can at least open the program 3-4 times out of 10.

### Expected results

The program should be stable!

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

non - happens with various projects

MariMakes, 2023-01-05:
Hey @B18rexracer,

Welcome to the Ultimaker Cura Github 🚀 
Sorry to hear you are having so much crashes 😖 

I'll have someone from the team take a look at your logs. 
Fingers crossed 🤞

casperlamboo, 2023-01-06:
Hey @B18rexracer,

Sorry to hear Cura is crashing so often. I took a look at the logs, but unfortunately could not really find any cause of the issue. Its a bit hard to resolve your issue, without knowing what exactly causes the issue. If you do recognize a pattern in the crashing and when you are able to get Cura in a state where it consistently crashes please do share these reproduction steps. Once we know how and when it crashes we can resolve the issue.

- @casperlamboo

casperlamboo, 2023-01-19:
Hi @za-x-az 

Thank you for the elaborate description. Could you maybe share your logs generated immediately after such a crash. That may help us alot! 

- @casperlamboo

za-x-az, 2023-01-19:
@casperlamboo 
Sure where do I find this log?
should be easy to force this crash/freeze up

MariMakes, 2023-01-20:
Hey @za-x-az,

You can find your log file here:
Windows: %APPDATA%\cura\5.2\cura.log 
or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

za-x-az, 2023-01-23:
@MariMakes @casperlamboo 
Is this good?
i removed the old log file so it should only contain a start-up + crash

[Forced_crash.log](https://github.com/Ultimaker/Cura/files/10482277/Forced_crash.log)

MariMakes, 2023-01-26:
Oh! If it's related to your second monitor... there are these two workarounds that might help you. 

1. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
2. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

I'll have someone look at your logs 🤞

MariMakes, 2023-03-31:
Quick update on our side,

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download link and discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)