Ultimaker/Cura #14178: SLicing Failed

TRC-Loop, 2022-12-31:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

1. Tried to slice a Low-Poly Model
2. Failed

### Actual results

An error Message appears that said Report this bug.

### Expected results

The Model should be sliced so i can save it as gcode

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

31.12.2022

GregValiant, 2023-01-03:
Thanks for the report bug without a project file and a Cura log file there can be no investigation.
To create a project file load the model, set Cura up to slice, and use the "File | Save Project" command.
To find the log file use the "Help | Show Configuration Folder" command.
Zip the two files and post the zip folder here.

TRC-Loop, 2023-01-03:
The project was empty