Ultimaker/Cura #14183: Odd IDEX move when starting print with tool 1

therealzoomgod, 2023-01-01:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom IDEX

### Reproduction steps

Load a part to print with tool 0, a raft to print using tool 1.  Print starts using tool 1

### Actual results

A move to 0,0 occurs which is the opposite side of the bed from tool 1 parking.  This can cause a move out of range when there is an offset aplied to align tool 1 with tool 0

G0 X0.00 Y0.00
;TIME_ELAPSED:522.022474
;LAYER:0
G1 F1200 E209.27738
G1 F600 Z2.01
G92 E0
park_extruder1
T0

### Expected results

I would expect tool 1 to be sent to the opposite side of the bed

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Gear Test.zip](https://github.com/Ultimaker/Cura/files/10329629/Gear.Test.zip)

therealzoomgod, 2023-01-01:
Forgot to add the G0 X0 X0 can be found just prior to park_extruder1 in gcode

GregValiant, 2023-01-05:
Thanks for the report.
I can see that "park_extruder1" is in your end gcode for extruder 1.  I don't understand where the "G0 X0.00 Y0.00" line is coming from.  It might be going in because that is the last time Extruder 1 is used.
Your Search and Replace workaround is a good one.  I assume that your printer will understand "park_extruder1".
Someone from the Cura team will take a look at this.

therealzoomgod, 2023-01-22:
Just a follow up as I ran into another odd piece of gcode when switching to T1, causes same issue,  No idea why it wants to move tool to the opposite side of bed when print is in the middle of the plate.  No idea how to filter that one out with post processing search and replace since the Z part is needed.  May have to overload G0 moves with a klipper macro and filter them.

T1
G92 E0
M104 T0 S175
M105
M109 S215
M106 S255
M104 T0 S195
G1 F1500 E-2
G0 F7200 X0.00 Y0.00 Z1.8   <---------------