Ultimaker/Cura #14185: Slicing failed on model with support blockers and per model density change

KD7UM, 2023-01-02:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

Made changes to the blocking support and slicing failed.  I am using per model infill change to 100% for the chimney and lower section of the smoke.

### Actual results

Slicing failed

### Expected results

Slicing should work.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The slicing indicator just keeps going back and forth.

[image]
[CE3E3V2_globe-new-trees-c.zip](https://github.com/Ultimaker/Cura/files/10329989/CE3E3V2_globe-new-trees-c.zip)

GregValiant, 2023-01-03:
Thanks for the report.
I didn't have any trouble slicing your project file so I'm unable to duplicate the problem.
If you select "All" the meshes and rotate them 90° on the Cura build plate you may get this to slice.  I'll mark this as a duplicate as there have been a lot of reports of good models that won't slice when in certain orientations on the Cura build plate.

Looking at the project - the  "lightning" infill pattern cannot be set to 100% density as by it's nature it is meant to be sparse.  You can change the Infill Pattern of that blocker to "Lines" and get the smoke to print with solid infill.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker