Ultimaker/Cura #14188: Fan bug with "minimum layer time" tool

jrdavidg, 2023-01-02:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

- Activate "minimum layer time" and set it to 10 seconds
- Activate "regular/maximum fan threshold" and set it to 0 seconds
- Activate "regular fan speed" and set it to 30%
- Activate "maximum fan speed" and set it to 100%
- Set fan speed to 30%
- Slice the model

### Actual results

Fan will run at maximum speed at Layer 45: M106 S255

### Expected results

Fan should be at 30%, without any kind of changes in the gcode.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[COOLING_TEST.zip](https://github.com/Ultimaker/Cura/files/10331459/COOLING_TEST.zip)

GregValiant, 2023-01-03:
Thanks for the report.
I followed your steps and indeed with those settings the fan goes to 100% at Layer 37 (with my print speed settings).  You have set the Maximum to 100% and requested no steps between regular and maximum so when the layer time fell to 10 seconds the fan kicked up higher.
That appears to me to be the way it's supposed to work.
In regards to "Regular/maximum fan speed threshold" the tooltip states that "layers that are slower will use regular fan speed and layers that are faster will use maximum fan speed".  That makes it sound like it is some sort of over-ride for "Minimum Layer Time" but that doesn't appear to be true.
Setting "Regular/Maximum...threshold" to 15seconds results in multiple steps between 30% and 100% fan usage.

It appears that Cura is working as expected but the Tooltip for the "Regular/Maximum Fan Speed threshold" could use some work.  Things like that happen when English isn't the first language of the software engineers.

Someone from the Cura team will take a look.

jrdavidg, 2023-01-03:
> Thanks for the report. I followed your steps and indeed with those settings the fan goes to 100% at Layer 37 (with my print speed settings). You have set the Maximum to 100% and requested no steps [...]

Thaks for the reply! Yesterday I tested Cura 4.13 (last version before 5.0 branch) with the same settings I had issues with 5.2.1 and that M106 S255 dindn't appear inside the gcode file. That's the reason I think that's a bug.