Ultimaker/Cura #14192: Differences normal supports Cura 5.2.1 vs 5.3.0 alpha+xmas

Murciaprint3D, 2023-01-02:
### Application Version

5.3.0

### Platform

Windows10

### Printer

Artillery Sidewinde X1

### Reproduction steps

Hello everyone, I have detected that there are differences in the NORMAL supports between the two versions of CURA 5.2.1 and 5.3.0 alpha+xmas .
With the same file and settings there is a difference in printing time and meters of filament.
Why is this happening?
Is there any change in this type of supports?

Regards

### Actual results

.

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-01-05:
Hey @Murciaprint3D,

Thanks for your report 👍
Wow! I'm super curious aswell why this is happening.

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

ThomasRahm, 2023-01-05:
Most likely because of https://github.com/Ultimaker/CuraEngine/pull/1671

closed 2023-01-20 by github-actions[bot]