Ultimaker/Cura #14193: Enable Print Cooling 5.2.1

pmhoppe, 2023-01-02:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Longer LK5 Pro updated ( 2 Cooling Fans )

### Reproduction steps

Enable Print Cooling is checked (enabled) 

### Actual results

The gcode always includes M107 after Start Gcodes section.

### Expected results

The cooling fans should turn on as directed in the Start gcode section.

Line 21 turn fans on
Line 27 turns fans off

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[1LLK5PRO_testdome.zip](https://github.com/Ultimaker/Cura/files/10332857/1LLK5PRO_testdome.zip)

Includes:
project file
gcode file

GregValiant, 2023-01-03:
Thanks for the report.
There seems to be a couple of things going on here.

- As your bug report states it does appears that there is no fan control for a raft and that fan control only starts with Layer:0.  So the M107 line does indeed shut the fan off and it does not restart during the raft.  The way you have your settings it appears that the fan doesn't start until Layer:3 in the gcode.  If you set an Initial Fan Speed then the fan will turn on for Layer:0.
- Adding "M106 S{cool_fan_speed_0} to the StartUp Gcode does not negate the presence of the M107 line.

- You say you have two fans but it appears to be configured as a single fan circuit maybe controlling two blowers (is that correct?).  You have extruder 1 configured to control Fan #0 which is the normal situation.  It doesn't matter how many blowers are on that circuit.  It might be important to know if there are both P0 and P1 fan circuits.

The Cura team will take a look.  In the meantime a workaround would be to use the Search and Replace post processor and Search for M107 and replace it with M106 S255 or whatever fan speed you want for Layer:-6 and the raft.

- The gcode file you provided was not created using the StartUp Gcode of the Longer printer that is present in the Project File.  We need to have consistent information in order to debug.
- The startup gcode of the printer within the project file is incorrect and won't work because the heating commands come after the purge lines.  That will cause a cold extrusion warning.  In addition - the purge lines along with a lot of the raft simply won't print as the heaters don't even begin to get warm until the purge line moves are complete.  The raft then starts with a cold bed and cold nozzle.  On a small print like that - you could possibly finish the whole thing without extruding any plastic.

pmhoppe, 2023-01-04:
> ### Application Version
> 5.2.1
>
> ### Platform
> Windows 11
>
> ### Printer
> Longer LK5 Pro updated ( 2 Cooling Fans )
>
> ### Reproduction steps
> Enable Print Cooling is checked (enabled)
>
> ### Actual results [...]

GregValiant, 2023-01-04:
@pmhoppe I'm seeing the same situation in your last post as in the original report.  This is the StartUp Gcode for the Longer printer that is active in the project file.  You can see that the M104 and M140 lines are **_AFTER_** the "Draw first Line" and "Draw second line" commands.  You are trying to extrude with a cold nozzle and cold bed.

; LONGER Start G-code
G21 ; metric values
G90 ; absolute positioning
M82 ; set extruder to absolute mode
;M107 ; start with the fan off
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
;M104=print temp, M140 = bed temp, 109 = print temp wait, 190 = bed temp wait
M104 S{material_print_temperature_layer_0} ;Start heating extruder ; pmh
M140 S{material_bed_temperature_layer_0} ;Start heating bed ; pmh
M104 S{material_print_temperature_layer_0}  ; pmh
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

This is the startup gcode from your included Gcode file:
; LONGER Start G-code
M190
M106 S255

pmhoppe, 2023-01-04:
Thanks for the reply. I am new to Cura and should have researched more before issuing a bug report. I had enabled cooling without setting any fan speeds. Once I had set a fan speed the M107 disappeared from my posted file. Definitely not a bug. Sorry for wasting you time. Paul

GregValiant, 2023-01-04:
Cura is full featured and industrial strength.  Among other things that means a steep learning curve.
I'll go ahead and close this then.

One piece of advice - fix your startup gcode or you are going to have issues.

closed 2023-01-04 by GregValiant