Ultimaker/Cura #14194: Ultimaker Cura, New printer

omaraljumaa722, 2023-01-02:
### Is your feature request related to a problem?

I got a new 3d resin printer but when I tried to add the printer to Cure I can not find it, it is AnyCubic Photon M3, could you guys please add this printer to the app so I can slice and print

### Describe the solution you'd like

adding the printer to the app 

### Describe alternatives you've considered

there is no alternatives in the Cura app, as of now I have to use other slicer to slice and print 

### Affected users and/or printers

AnyCubic Photon M3

GregValiant, 2023-01-03:
Printer definition files come from the community or the manufacturer of the printer.  UltiMaker accepts printer definitions for other manufacturers printers but can't support them directly.  You may read up on adding your printer [HERE](https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura).

Asterchades, 2023-01-04:
OP isn't talking about adding just a new profile, Greg - they want to add an entire new slicing method. Their printer is a resin-based MSLA, not filament, so there's exactly zero existing support in Cura beyond the concept of "layers".

Pretty sure that's beyond the scope of the Cura project given that Ultimaker doesn't make or sell resin printers. But I'll defer to the devs for clarification on that one.

GregValiant, 2023-01-04:
Thanks @Asterchades.  I do seem to recall _Nallath_ saying that they aren't supported.
I did read (somewhere) that at least a couple of resin printer owners were using Cura, but maybe they had some sort of post-processor to do a translation.