Ultimaker/Cura #14195: Flashforge Creator Pro/Pro 2 IDEX Support

skruglov2023, 2023-01-03:
### Is your feature request related to a problem?

It's frustrating to have to use multiple different slicers for different printers.

### Describe the solution you'd like

If cura had flashforge creator pro/pro 2 support, I wouldn't have to use the flashprint app for slicing any print variants, whether left/right extruder, mirror or duplicate mode for prints.

### Describe alternatives you've considered

Using the Flashforge FlashPrint app, which is severly lacking in functionality compared to Cura. 

### Affected users and/or printers

Flashforge printer users, except the dreamer, for which a profile exists

GregValiant, 2023-01-03:
Printer definition files are provided by members of the community or by the printer manufacturer.  The definition files are submitted to Ultimaker via Pull Requests for inclusion in Cura.  It would be impossible for Ultimaker to provide support for other manufacturers printers.
The Raise3D IDEX printers can be used with Cura as they have specific commands that can be entered in the StartUp Gcode to set the printer to Mirror, Duplicate, or Normal.  Users need to install the printer 3 times - one for each mode - and then select the printer according to the mode they intend to use.  The StartUp gcode for each installed printer would contain either M605 S0 (normal), M605 S1 (Mirror) or M605 S2 (duplicate).
If the IDEX printer requires the use of different axes (U, V, W) then Cura won't work.

x40-Community, 2023-01-13:
I think our Cura Idex Plugin project might be interesting for you. However, without a donation to the project, your printer will not be included and will not work with it.

We need the donations to pay the programmer. We couldn't find anyone who would otherwise implement the project.

Project Homepage
http://www.x40-community.org/index.php/9-cura-workflow/85-cura-idex-plugins

Demo Video
https://www.youtube.com/watch?v=Yzv-GpHitxI

AJKnabenbauer, 2023-03-05:
Not sure how helpful it is but I have a Creator 3 pro that I was able to get working in cura (right extruder only for now). I am not sure if the creator pro/pro 2 uses the same flavor of g-code though. **The Creator 3 pro specifically requires a big binary blob at the start of the gcode file that has the IDEX mode settings in it, along with the thumbnail image and some other bit-flags for settings.** So it is not possible to easily select the mode in the startup g-code. Attached is what I did to make cura work using only built in pre-processor scripts and start/end gcode.
[Flash Forge Creator 3 Pro Cura settings.pdf](https://github.com/Ultimaker/Cura/files/10893234/Flash.Forge.Creator.3.Pro.Cura.settings.pdf)

After slicing I am able to print the .gcode file from a flashdrive, or I can send it to the printer via flashprint. 

If you need help creating the custom printer in cura follow along to this guide from flashforge support (note I have not used the cura2ff software they show in the guide). https://docs.google.com/document/d/1_nZbFMvi8aj3wwPgVVtqZGOjrdIPiz7e/edit

I am working on getting the other extruder modes working but it is a huge PITA since flashforge doesn't publish their g-code format. I bought a simplify3d license since I was so sick of flashprint, and it has profiles for all the IDEX flashforge printers. So I pretty much started from what they did and trial and errored my way through the rest. 

If you need an alternative to flashprint, that works out of the box, try looking into simplify3d. It isn't cheap ($199) but is the only slicer I know of that has ready to go profiles for all the creator printers. Here is a screenshot of all the printers that have built in profiles. Word of caution though - simplify3d isn't nearly as customizable as cura. So if you like to fine tune all the little obscure settings of a slice profile you will probably be disappointed. Which is why I am back to cura.
[image]