Ultimaker/Cura #14197: slicing problem at cura 5.2.1

samedocalan, 2023-01-03:
### Application Version

5.2.1

### Platform

windows 11

### Printer

creality ender 3 v2

### Reproduction steps

when ı slicing item there is a problem on screen
[image]

### Actual results

ı came here

### Expected results

ı want to print my 3d firefighters model at cura 5.2.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

please solve my problem urgent

GregValiant, 2023-01-04:
Thanks for the report but without a log file and project file there can be no investigation as there is no information to go on.
To create a project file use the "File | Save Project" command.  To find the Cura.log file use the "Help | Show Configuration Folder" command.  Add those files to a zip folder and post the zip folder here.

gallegogti, 2023-01-04:
Hi, I had a similar problem. Attaching the log files.
[cura.log.1.zip](https://github.com/Ultimaker/Cura/files/10344797/cura.log.1.zip)

GregValiant, 2023-01-04:
I have found that well over half the time these slicing issues are caused by errors in the model.  Without the model to check, and a Project File that shows how the slice was set up, troubleshooting is tough.
I'm not a member of the Cura team so log files are mostly Greek to me.  (I can tell a Larkin pipe when I see one though.)
Have you tried rotating and/or moving the model on the Cura build plate?  There is a persistent bug on Windows installations that can keep an error-free model from slicing when it's at the midpoint of the build plate.

closed 2023-01-20 by github-actions[bot]

MariMakes, 2023-02-23:
Hey here,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.