Ultimaker/Cura #14199: Cura Crash when printing

nseaman5, 2023-01-03:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker s5

### Reproduction steps

1. Sliced 3D print file. 
2. Go to print and/ or save file and Cura crashes. I am unable to save any files or print anything either. 

### Actual results

Cura Crashes and says fatal error occured. 

### Expected results

Cura Prints or saves the designated file. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cannot upload any files since cura crashes when i attempt to save any files

MariMakes, 2023-01-05:
Hey @nseaman5,

Welcome to the Ultimaker Cura Github 🚀 
Sorry to hear you are blocked from printing 😞 

Do you know if Cura is crashing on this specific model or have you never had a working Cura? 
Can we have your configuration folder to see what might be preventing you from?

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

nseaman5, 2023-01-05:
> ---aFe WARNING: This email is from an EXTERNAL sender. Never trust attachments or links in this email message.---

Hello,

I was able to fix this problem. It had something to deal with the profile I had created. Not sure why it was bugging out but when I switched profiles the problem went away.

Thank you,
Nate.

MariMakes, 2023-01-06:
Hey @nseaman5,

Thanks for getting back to us. 
I'm glad your issue was resolved. 
I'll close the ticket.

closed 2023-01-06 by MariMakes