Ultimaker/Cura #14200: Bridge settings not applied to Extra Skin Wall on top of sparse infill

Bastel-Bodo, 2023-01-03:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom FFF printer (Original Prusa MINI+)

### Reproduction steps

I sliced a model in Cura that has a solid body and a cable canal carved into its top. To improve the top surface where the canal starts on the infill, I set **Extra Skin Wall Count** to 1. In the **Bridge Settings** section, I set the **Bridge Skin Speed** to a slower speed than regular top/bottom and **Bridge Sparse Infill Max Density** to a value greater than my selected infill density.

### Actual results

In the resulting G-Code, the fill lines of the top surface on the sparse infill were printed using the reduced bridge skin speed, but the extra skin wall around them was still printed at the faster top/bottom speed. This caused some parts of the extra wall to fail, thus killing the "anchor" for the fill lines of the top surface.

Start of the extra skin wall:
[image]

Start of top skin surface fill lines:
[image]

### Expected results

The extra skin wall should be printed at the same speed as the top skin surface it surrounds when the top surface skin is printed on sparse infill at a reduced speed via the Bridge Settings, in order to ensure a reliable anchor for the fill lines of the top surface.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

You can observe the described behavior in layers 13 and 16 of the sliced result of the attached project, where top surface skin patches are printed on top of sparse infill. For the extra skin wall, feed rate is set to 4200, whereas for the fill lines of the top skin surface, it is set to 1500.

[LightHolder-3mf-gcode.zip](https://github.com/Ultimaker/Cura/files/10339055/LightHolder-3mf-gcode.zip)

MariMakes, 2023-01-09:
Hey @Bastel-Bodo,

Thanks for your report 👍

This happens to be already reported internally and has a ticket on the backlog with the intent to improve. For internal reference CURA-9521. You are the first one to report it on Github. 
Thank you for that 😊 

Looking at your project file I think your model might benefit from some Skin Edge Support Layers.
If I understand correctly they might solve your issue with the sparse infill density. 
[image]

Bastel-Bodo, 2023-01-09:
Hi @MariMakes,
cool, that's good to know.
And also thanks for the suggestion regarding Skin Edge Support Layers - will definitely give this a try!

saumyaj3, 2023-10-06:
The issue is fixed in 5.5. Closing the ticket now.