Ultimaker/Cura #14204: Custom name when saving to removable drive

TheDeejster, 2023-01-04:
### Is your feature request related to a problem?

not related to a problem per se

### Describe the solution you'd like

It would be really nice if after slicing, above the "save to removable drive" button, there was a little text box wherein I could enter the name I want the exported gcode to have. I often replace models or need to print multiple of the same model with slight differences, and when I have either of those scenarios, I have to save the gcode, then manually enter the drive and change the name of the file before saving the next one. I feel as though this defeats the convenience of the save to removable drive button.

### Describe alternatives you've considered

Any alternatives just require more steps than I believe necessary. 

### Affected users and/or printers

Really anyone using Cura, but especially those who print variations of the same models.

### Additional information & file uploads

In case my description wasn't clear, I created a mockup design to portray my suggestion more clearly.
[image]
[image]

bumboo001, 2023-01-04:
Hello @TheDeejster ,
this feature already exist. You can change name of file you are editing under object list(same name will be saved on removable drive)
[image]

TheDeejster, 2023-01-04:
I was unaware that that was the case, thank you for clearing that up! I do
believe my suggestion would still be an improvement, as the function would
be clearer/easier to find, but I am glad that the feature exists and I
appreciate the reply. Thank you!