Ultimaker/Cura #14208: Tree supports - do not avoid support blocker

evilC, 2023-01-04:
### Is your feature request related to a problem?

The new tree supports are great, but not creating supports inside the support blocker area is not always desirable.

Consider the following slice:
[image]
Here, the intent is to not generate supports for the bridges inside the support blocker (The tops of the 6 vent holes on each side), but an undesired effect is also happening - the supports for the clips at the top of the picture are routing around the support blocker, when they don't need to

Admittedly, the solution here is rather simple (Just scale down the support blocker a bit), but that's more hassle, and may not always be possible?

### Describe the solution you'd like

An option that specifies that supports may travel THROUGH a support blocker, but may not support a face WITHIN the support blocker

### Describe alternatives you've considered

Size down the support blocker or use multiple smaller ones

### Affected users and/or printers

All

ThomasRahm, 2023-01-05:
Currently this behavior is intentional and permanently enabled. Making this into a setting would be very easy. 
The big question (for the guys at UltiMaker) is whether such a setting is needed (resizing the support blocker would achieve the same result here) or only causes setting-bloat.

evilC, 2023-01-05:
Right, but consider geometry such as this:
[image]
(Section analysis enabled so that you can see inside)
If this option were added, you would need one support blocker to stop it trying to create supports for the tops of the feet (They are bridges, so would not need support)
With the code as it stands, you would need 4 support blockers

MariMakes, 2023-01-16:
Hey @evilC,

Thanks for raising this. 
I really like your example. 👍 
I've brought it up for consideration to the team. 

For internal reference - CURA-10161

rgreer4, 2023-05-09:
For what it's worth, having the ability to block where supports route and merge, and thus control/guide them a little, is a quite useful feature I wouldn't want to see removed.  (You could still use actual models set to have no walls printed I suppose as blockers -- as you have to do that to control/break up the tree monoliths in the prior incarnations of tree supports, and it's not ideal or obvious.)

I'd vote for a checkbox or similar on the support blocker object, letting you pick which behavior you want.  Both can be desirable.

MariMakes, 2023-05-09:
Update from our side. @ThomasRahm seems to have the resolved this issue here: https://github.com/Ultimaker/CuraEngine/commit/1d6203837a23d8c7fba1925622f27e88e305672e
You can expect it in the next Cura release 😉 👍