Ultimaker/Cura #14215: Unexpected error slicing jpg

sandlerr, 2023-01-05:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Opened a JPG file and hit slice. The image is 3855x2481 px. The "Convert Image" settings are defaults except for Smoothing set to zero. Both color models trigger the same problem.

### Actual results

An unexpected error occurs, suggesting a bug report be created.

### Expected results

I'd like the image to be sliced.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

```
[warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,506 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.506] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,513 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.514] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,514 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.514] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,515 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.514] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,516 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.516] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,518 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.518] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,523 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.523] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,524 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.523] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,524 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.523] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,526 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.526] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,527 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.526] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,527 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 05:36:43.527] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 05:36:43,589 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221225477!
2023-01-06 05:36:43,591 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
2023-01-06 05:36:43,591 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [366]: Killing engine process
2023-01-06 05:36:43,592 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [369]: Engine process is killed. Received return code 3221225477
```

GregValiant, 2023-01-06:
_"Backend exited abnormally with return code 3221225477!"_
That line has shown up here many times.  I'll mark this as a duplicate and someone from the Cura team will take a look.
In the meantime, have you tried moving and/or rotating the model on the Cura build plate?  It may be helpful to know if that allowed the model to slice.

closed 2023-01-21 by github-actions[bot]

MariMakes, 2023-02-23:
Hey @sandlerr,

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.