Ultimaker/Cura #14216: More Control Over Draft and Ooze Shields

rachael7, 2023-01-05:
### Is your feature request related to a problem?

I have ongoing intermittent struggles with draft and ooze shields.  Many of my models are dual material and the alternating materials in the shield, combined with the thin, fragile nature of the shield, lead to more failures than I would like. It's very frustrating to nail the settings on a complex print, only to have the print fail because the draft shield collapsed into the part or detached from the build plate.

### Describe the solution you'd like

I would like more control over the draft and ooze shield.  Ideally, Cura would insert the shield as another part on the build plate and we could make all the adjustments to it that we make to any part, but at a minimum, it would be extremely helpful if in addition to the existing settings, we also had the following settings:

- Shield Thickness. Often the difficulty could be eliminated simply by having a shield that was two line widths thick instead of only one.
- Shield Material. Having the choice of Extruder 1, Extruder 2, and Alternating would be excellent.
- Shield Speed. Acceleration and Jerk are less important, but could also be included.
- Shield Flow Rate
- Shield Line Width

### Describe alternatives you've considered

My usual workaround is to design the draft/ooze shield in my solid modeling software and add it to the build plate as a separate part. This is very time consuming though, and unnecessary when the existing algorithms generate the shield so easily.

### Affected users and/or printers

Draft shields and ooze shields are not employed by most users, but they are a powerful tool in primarily two use cases: People who do multiple extruder or multiple material prints will benefit a lot, as will those who use high temp materials and wish to maintain a warm environment around the printed part (to improve layer bonding, etc). With the number of new multi-material printers coming out such as the Bambu Labs machines, the increasing popularity of two-into-one hotends, and IDEX printers, and the ever-increasing array of high temp engineering filaments, it is likely that more people will begin to use the shields and experience the sorts of problems I have.

GregValiant, 2023-01-06:
I just wanted to comment that you have provided a concise, well thought out feature request.  Thank you.

Rouv, 2023-03-24:
I would be very happy about the implementation too! 
Especially the ability to set the thickness and only adding a brim to the shield would be very helpful in my case.