Ultimaker/Cura #14217: Weird slice error

Shakansis, 2023-01-06:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender 3 V2

### Reproduction steps

1. import the .stl file
2. click on the slice button

### Actual results

The slice begins and with approximately 30% it returns with an error message.

### Expected results

Slice complete and I am able to print the GCODE

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10356466/cura.log)

ThomasRahm, 2023-01-06:
This part of the log suggests to me that this crash could maybe come from the tree-support code for some reason:
`2023-01-05 21:17:34,270 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-05 21:17:34.270] [info] Processing support tree mesh group 1 of 1 containing 1 meshes.`
`2023-01-05 21:17:37,395 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221226505!`
Does it also crash if you disable support?
If no: Does it crash again if you try again? If yes: Is there any chance that you can include the project file, and if you can, tell me how many cores/logical processors (seen in the task manager in the performance tab) your cpu has?

If your issue was the slice attempt 3 minutes later at `2023-01-05 21:20:18,507`, disregard the above.
You can try moving the model around a bit to hope that the bug does not occur.

ThomasRahm, 2023-01-06:
I think this could be a duplicate of #14113 where there is an project file with a reproducible crash

Shakansis, 2023-01-09:
Hello Thomas, 

Thank you for your response :)

Yes, it looks like it is related to the new tree support and, yes, I think it is a duplicate of https://github.com/Ultimaker/Cura/issues/14113, sorry...

But, the answers to your questions:

Cores
[image]

The project file is 49MB
[image]

But you can find the .stl here:
https://www.printables.com/model/343261-balrog-figure-lord-of-the-rings

It had the tree support turned on, 0.16mm quality, brim turned on, retraction (7mm), and coasting turned on.  
Material: PLA
Temps:
200C - hot end
60C - Table

Regards,
R.

MariMakes, 2023-01-09:
Hey @Shakansis,

Thanks for your report 👍

We would love to add your file as a testcase. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
For big files, you may need to use WeTransfer or similar file-sharing sites.

closed 2023-01-23 by github-actions[bot]