Ultimaker/Cura #14218: wont let me slice a print

Ealbright88, 2023-01-06:
### Application Version

5.2.1

### Platform

windows 64

### Printer

ender 5 pro

### Reproduction steps

1. opened print file, one i had previous loaded before
2. change the X, Y, and Z to meet my needs
3. hit slice 

[Glove_box_holder_for_standard_pegboard_4982937.zip](https://github.com/Ultimaker/Cura/files/10356439/Glove_box_holder_for_standard_pegboard_4982937.zip)

### Actual results

1. cura says it encountered a bug and to report it

### Expected results

for the print to slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

1. I restarted the computer
2. i changed the X, Y, and Z back to the original and it sliced the print
3. still ran into the same bug when i changed it to the size i needed

Beornhelm, 2023-01-06:
The same thing has happened for me as well. I went to slice a large file with fairly standard settings, and it told me about a fifth of the way through that it encountered an unexpected bug and that the slicing had failed.

DerGenaue, 2023-01-06:
Can you try just moving the object a bit around on the buildplate and re-trying the slice?

Beornhelm, 2023-01-06:
Oh yeah, I did try that a bunch to no effect. I did manage to get it to
work once by changing my infill from 8% to 10%.

MariMakes, 2023-01-09:
Hey @Ealbright88,

Thanks for your report 👍

I checked the file that you shared and I'm getting an error that the mesh is not watertight. 
[image]

Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

If that doesn't resolve your issue we need some extra information because the files slices for me like any other. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Ealbright88, 2023-01-23:
Your response fixed the issue

Sent from my Verizon, Samsung Galaxy smartphone

Closed #14218<https://github.com/Ultimaker/Cura/issues/14218> as completed.

MariMakes, 2023-01-26:
Thanks @Ealbright88,

I'm glad your issue was fixed.  😊 
I'll close this issue.

closed 2023-01-26 by MariMakes