Ultimaker/Cura #14221: Cura not starting on Kubuntu 22.04 with Creality Ender-6 selected and output: Cyclic dependency on Ultimaker-Cura-5.2.1-linux-modern.AppImage and Ultimaker-Cura-5.2.1-linux.AppImage

Elektronikus, 2023-01-06:
### Application Version

5.2.1-linux

### Platform

Kubuntu 22.04 LTS

### Printer

Creality Ender-6

### Reproduction steps

1. open cli
2. starting  ./Ultimaker-Cura-5.2.1-linux-modern.AppImage or ./Ultimaker-Cura-5.2.1-linux.AppImage 
3. cli freezes for a long time
4. startup image appears and freezes at 
[image]

### Actual results

PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaKmeCSj/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaKmeCSj/share/cura/resources/qml/Actions.qml"
----------------------------------

### Expected results

Just startup

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

cura.cfg: renamed to txt to uplodad
[cura_with_Ender_6.txt](https://github.com/Ultimaker/Cura/files/10359262/cura_with_Ender_6.txt)
[stderr_not_starting.log](https://github.com/Ultimaker/Cura/files/10359263/stderr_not_starting.log)

If I remove:
[cura]
active_machine = Creality Ender-6
recent_files = /home/ehinger/3D-Druck/Tastaturaufsteller-Body.3mf;/home/ehinger/3D-Druck/Counter/GPSEFC_EnclosureV2.stl;

Cura starts! but it reports a lot of problems on the cli:
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///tmp/.mount_UltimaGDMa05/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"

MariMakes, 2023-01-09:
Hey @Elektronikus,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Thanks for trying both versions of our AppImage. I'm sorry to see that you are stuck. 😖 
This report is outside my area of expertise so I'll have to bring it up to the team to see what they can do. 

Fingers crossed 🤞

Elektronikus, 2023-01-09:
Hey MariMakes, 
thank you for the response. All the messages are around the property width. After removing 
[cura]
active_machine = Creality Ender-6
recent_files = /home/ehinger/3D-Druck/Tastaturaufsteller-Body.3mf;/home/ehinger/3D-Druck/Counter/GPSEFC_EnclosureV2.stl;
and selecting an ultimaker printer, the SW is starting 
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaBD35Gs/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaBD35Gs/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
If I select than the Ender 6 the SW is accepting the printer but freezing after some zooming in and out.
On Windows 10 the Ender 6 is working fine

With best regards

MariMakes, 2023-05-01:
Hey @Elektronikus,

Wow, I'm so sorry, this issue has gotten lost in our pile. 
Are you still stuck? 

If 5.3.1 does not resolve it for you, do you mind checking a nightly build with an updated QT version? 
You can find [the download link here.](https://github.com/Ultimaker/Cura/discussions/15098)

If it doesn't resolve your issue, please let us know as well then we'll take another look.

closed 2023-05-30 by github-actions[bot]