Ultimaker/Cura #14225: Slicing failed with an unexpected error

nebukadnet, 2023-01-06:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Tronxy X5SA

### Reproduction steps

I put in any model into cura, like a benchy
I click slice.

### Actual results

The bar doesn't progress, after a few seconds it tells me:
"Slicing Failed
Slicing failed with an upexpected error. Please consider reporting a bug on our issue tracker."

### Expected results

Cura should slice the model

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/10361952/stderr.log)

MariMakes, 2023-01-09:
Hey @nebukadnet,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Is it possible that your printjob is located at 0, 0, 0 and it does slice if you move it a little away from the center of the buildplate? Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.

Do you have a project file for us? It contains the printer and settings we need for troubleshooting.
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.

nebukadnet, 2023-01-09:
Hi, Thanks for responding!I did try that.I solved the issue by restarting my computer. Restarting cura wasn’t enough, I had to fully restart my computer, but the issue was solved and hasn’t returned since. I’m sorry to say I didn’t create a project file at the time. Now it works, so I could give you a project file, but I don’t think it would help you. Kind regards

MariMakes, 2023-01-09:
Hey @nebukadnet,

Thanks for getting back to us. 
If your issue has been resolved, I'll close the issue 👍 

Happy Printing!

closed 2023-01-09 by MariMakes