Ultimaker/Cura #14228: Bug

mrmidniteowl, 2023-01-06:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

Sliced a job - click on 'Report a bug' and here I am, no log, no real definition of the error?

### Actual results

Error

### Expected results

It should slice

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10363072/cura.log)

2023-01-06 14:55:13,577 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [89]: Adding QT6 plugin path: C:\Ultimaker Cura 5.2.1\PyQt6\plugins
2023-01-06 14:55:13,578 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [99]: Adding QT5 plugin path: C:\Ultimaker Cura 5.2.1\PyQt6\plugins
2023-01-06 14:55:13,578 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing Ultimaker Cura
2023-01-06 14:55:13,578 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.2.1
2023-01-06 14:55:13,578 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.2.0
2023-01-06 14:55:13,578 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None
2023-01-06 14:55:13,578 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\lib\uranium must be a folder.
2023-01-06 14:55:13,578 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\lib64\uranium must be a folder.
2023-01-06 14:55:13,579 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\lib32\uranium must be a folder.
2023-01-06 14:55:13,579 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Ultimaker Cura 5.2.1\plugins must be a folder.
2023-01-06 14:55:13,579 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Resources\uranium\plugins must be a folder.
2023-01-06 14:55:13,579 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\Resources\cura\plugins must be a folder.
2023-01-06 14:55:13,579 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
2023-01-06 14:55:13,579 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is C:\Users\jstevens\AppData\Roaming\cura\5.2
2023-01-06 14:55:13,579 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is C:\Users\jstevens\AppData\Roaming\cura\5.2
2023-01-06 14:55:13,579 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is C:\Users\jstevens\AppData\Local\cura\5.2\cache
2023-01-06 14:55:13,579 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Ultimaker Cura 5.2.1\share\uranium\resources\bundled_packages\uranium.json
2023-01-06 14:55:13,579 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Ultimaker Cura 5.2.1\share\cura\resources\bundled_packages\cura.json
2023-01-06 14:55:13,579 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2023-01-06 14:55:13,579 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
2023-01-06 14:55:13,580 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Ultimaker Cura 5.2.1\share\uranium\resources\bundled_packages\uranium.json
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Ultimaker Cura 5.2.1\share\cura\resources\bundled_packages\cura.json
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.PackageManager._loadManagementData [201]: Loaded user packages management file from C:\Users\jstevens\AppData\Roaming\cura\5.2\packages.json
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 
2023-01-06 14:55:13,580 - INFO - [MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [121]: Loading plugin configuration file 'C:\Users\jstevens\AppData\Roaming\cura\5.2\plugins.json'
2023-01-06 14:55:13,580 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location C:\lib\cura must be a folder.
2023-01-06 14:55:13,581 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin FileLogger 1.0.1
2023-01-06 14:55:13,588 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin LocalContainerProvider 1.0.1
2023-01-06 14:55:13,594 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin LocalFileOutputDevice 1.0.1
2023-01-06 14:55:13,599 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin CameraTool 1.0.1
2023-01-06 14:55:13,605 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin MirrorTool 1.0.1
2023-01-06 14:55:13,613 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin RotateTool 1.0.1
2023-01-06 14:55:13,620 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin ScaleTool 1.0.1
2023-01-06 14:55:13,624 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin SelectionTool 1.0.1
[... 195 lines cut ...]
2023-01-06 14:56:38,030 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:38.029] [info] Progress: layerparts accomplished in 0.07799999999997453s
2023-01-06 14:56:38,031 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:38.029] [info] Starting inset+skin...
2023-01-06 14:56:39,921 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:39.920] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:39,924 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:39.924] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:39,956 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:39.957] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:39,988 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:39.988] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,029 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.029] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,030 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.029] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,100 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.100] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,200 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.200] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,226 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.226] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,245 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.245] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-01-06 14:56:40,257 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:40.257] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
[... cut ...]
2023-01-06 14:56:41,511 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [115]: [Backend] Calling engine with: ['C:\\Ultimaker Cura 5.2.1\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2023-01-06 14:56:41,538 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-01-06 14:56:41.538] [info] Connecting to 127.0.0.1:49674
2023-01-06 14:56:41,540 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Connected
2023-01-06 14:56:41,540 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [185]: Backend connected on port 49674

MariMakes, 2023-01-09:
Hey @mrmidniteowl,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

mrmidniteowl, 2023-01-09:
Adjusting the location did the trick!

Thanks you,
Jeff Stevens

MariMakes, 2023-01-10:
Hey @mrmidniteowl,

Thanks for getting back to us.

We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13643

Good luck with your printjob. 🎉

closed 2023-01-10 by MariMakes