Ultimaker/Cura #14233: Slicing Failed

MateMar04, 2023-01-07:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Hellbot Magna 2 230

### Reproduction steps

1. Downloaded a .stl file and opened it with Ultimaker Cura 5.3.0
2. Tried to slice it but it won't

### Actual results

Keeps trying to slice bur without good results

### Expected results

Slice

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Screenshot in prepare mode: [image]
Screenshot in preview mode: [image]
[HM2230_Dowel.zip](https://github.com/Ultimaker/Cura/files/10366571/HM2230_Dowel.zip)

MateMar04, 2023-01-07:
If I lift the object, so it can rest on a support, it slices.
[image]

GregValiant, 2023-01-07:
Thanks for the report.
There is a persistent bug that has been reported numerous times that can keep a good model from slicing when it is in certain positions on the build plate.  Your project file sliced for me on the first try.
I had never thought of moving a model up, but moving it in the X Y or rotating the model will often fool the bug.
This bug is being worked on and the Cura team has numerous examples of this problem so I'll mark this as a duplicate and close this report.  I hope you understand.

closed 2023-01-07 by GregValiant

MateMar04, 2023-01-08:
I "solved" it duplicating the model and then deleting the original model

GregValiant, 2023-01-09:
I'll try that on the next one of these reports that come through (there have been a lot).