Ultimaker/Cura #14234: Filament sensor error not detected and continued to print without filament when Print via USB

ThisLimn0, 2023-01-07:
### Application Version

5.2.1

### Platform

Windows 10 (64Bit)

### Printer

Anycubic Kobra Max

### Reproduction steps

1. Started an overnight print with the rest of a 500g spool of filament.
2. Refill was scheduled in the morning
3. Printer showed an error on the screen but did not stop printing.
4. 

### Actual results

Printer prints without filament into the air.

### Expected results

Printer printed without filament into the air.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10366587/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/10366589/stderr.log)

GregValiant, 2023-01-07:
Thanks for the report.
I'm not sure how this could be a Cura bug.  Cura created the Gcode file and you used the USB plugin to send it to the printer.  It sounds like that worked as expected.

The filament sensor is part of the printer and it would seem the print should have been suspended by the printer processor when the filament sensor sent the "runout" signal.  Instead the printer seems to have ignored the filament sensor and kept processing the gcode commands OR the signal was never sent.  It's possible you have a glitch in your firmware or a problem with the sensor.

There is no ability in the USB plugin to interpret error codes that might come in from a printer.  They would likely vary from one brand of printer to another brand of printer and at any rate such functionality does not exist in the plugin.

I'll leave the bug label on this for now so you can respond here.  This sure sounds like a printer problem though.

GregValiant, 2023-01-09:
After due thought I've decided to remove the bug label and close this.  It is something that just can't be controlled from the server or gcode side and should have been handled by the printer itself.  I hope you understand.

closed 2023-01-09 by GregValiant