Ultimaker/Cura #14236: (Multi-material) Features from layer 1 show up in layer 5 but not 2, 3 or 4

evilC, 2023-01-07:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Snapmaker J1

### Reproduction steps

1. Open one of the files below (Ideally the cura project as everything is laid out) and align the models.  
Orientate them such as the face which has the text is touching the bed, and assign one model to Extruder 1, the other model to Extruder 2
2. Layer 1 should look like this
[image]
3. Scroll to layer 2, 3 or 4 - they all look as one would expect, such as this:
[image]

### Actual results

Scroll to layer 5 - BUG! It looks like this:
[image]
Notice how elements of the text from layer 1 are showing up on this layer, even though they did not show up on layers 2, 3 or 4  
It's also worth noting that I have a support blocker in place here (In the area where the blue lines are in this pic, but the issue repros without that)

### Expected results

Layer 5 should look basically the same as layers 2, 3 and 4 - the layer should basically be solid infill, or, depending on infill settings, it should be infill.  
I see no scenario where the detail from layer 1 should show up in layer 5 but not layers 2, 3 or 4

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Original two part model: [Riot shield 03.zip](https://github.com/Ultimaker/Cura/files/10367040/Riot.shield.03.zip)
Cura project: [Riot shield 03 Cura.zip](https://github.com/Ultimaker/Cura/files/10367042/Riot.shield.03.Cura.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10367077/cura.log)

MariMakes, 2023-01-09:
Hey @evilC,

Thanks for your report 👍
That's a super tricky detailed model. 
What do you use it for?

I don't have your custom 3D printer in my Cura, so I troubleshooted on an Ultimaker S5 instead. 
[image]

I believe it might be a slight alignment issue. 
[image]
I believe I've been able to align your model correctly here:
[RiotShield.zip](https://github.com/Ultimaker/Cura/files/10373584/RiotShield.zip)

If that's not it, can you doublecheck if your mesh modifier is assigned to the red extruder?

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for.  Would you still like to keep this issue open or do you agree that we can close it?

GregValiant, 2023-01-09:
I agree with @MariMakes that this is expected behavior.
On layer 1 a few areas start out with "bottom layer skin" and others start out with "Walls".  Layer 4 satisfies the Bottom Layer Count in some areas and so the Infill for those specific areas starts on Layer 5.  Other areas that started out with Walls on layer 1 need one more layer of skin to satisfy the "Bottom Layer Count" and it goes down on layer 5.  By layer 6 the Bottom Layer Count has been satisfied and the infill goes down across the model.
The clue is that the wide area across the bottom is straight on one side but follows the lettering "walls" on the other.
So - not a BUG!  Sorry @evilC I couldn't resist.  I had a Gru moment there.

evilC, 2023-01-09:
@MariMakes
> What do you use it for?
It goes on the front of a drone

It serves three purposes:
1) Protects the battery from frontal impacts
2) Complies me with UK legislation that says I must have my "Operator ID" (Uniquely identifies me to the authorities) on any drone that I fly.
3) Uniquely identifies that drone amongst my others (ie the `03` part lets me know that it is drone 3 - I have 4 identical drones, so it makes it easy to work out which is which)

Could you maybe tell me what exactly you did to align it perfectly? I am assuming that the offset I used (Y was set to -0.1 in mine I believe) was slightly off. Just having a copy of this file aligned perfectly does not help me much, as I have to do the same for drones 1, 2 and 4

@GregValiant OK, I see. It's not really an issue, I just thought I would report some oddness that I saw. Feel free to close

GregValiant, 2023-01-10:
I'll go ahead and close this then.  Good luck with the drones!

closed 2023-01-10 by GregValiant