Ultimaker/Cura #14237: "Align face to build plate" does not work for grouped or merged models

evilC, 2023-01-07:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Snapmaker J1

### Reproduction steps

1. Import attached 3MF model which comprises of two parts
2. Merge or Group the two parts
3. Select the merged / grouped model
4. Select Rotate
5. Click `Select face to align to the build plate`
6. Click the front face with the text on it

### Actual results

Model is de-selected and does not change orientation

### Expected results

Model should be rotated so that the selected face touches the build plate

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10367122/cura.log)
[Riot shield 03.zip](https://github.com/Ultimaker/Cura/files/10367124/Riot.shield.03.zip)

MariMakes, 2023-01-09:
Hey @evilC,

Thanks for your report 👍
What an interesting case 🤔 

It seems that aligning the groups also doesn't work in Cura 4.13, so it hasn't worked in quite a while.
I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. 

Fingers crossed 🤞

evilC, 2023-01-10:
@MariMakes 
This is quite a big issue for owners of IDEX printers.  
Presently, I see no way to accurately align a multi-part print AND orientate it to any face.
If, for example, with the attached file, you center both of the models (Without grouping / merging), and use the `Select face to align to the build plate` option, it DOES NOT align them as they were originally aligned. It is close, but not perfect.  
The only way as far as I know is to manually tweak the X/Y position and this takes a long time.

For example, if I do not group the two parts of the model, align them to the face I want to print on, and center both models, this is what I get: (Notice how the text is not perfectly aligned - the blue text is too low down)
[image]

Here, I have moved one of the models by 0.1mm on the build plate - note how it is now properly aligned:
[image]

Now as far as I am aware, if you have multiple models in a 3MF file, they all share the same coordinate space, so there should be zero reason why it cannot work it out. The problem is clearly not solely limited to the `align to face feature`  

Here is a version of the model that I pre-orientated within my 3D modelling software so that the face I want to print on is already aligned to the build plate:  
[Riot shield 03 orientated.zip](https://github.com/Ultimaker/Cura/files/10383839/Riot.shield.03.orientated.zip)
However, when you import this model and center both of the models, the text is STILL not aligned correctly.  
If you select both models and `Group` them, it aligns them incorrectly.  
If you select both models and `Merge` them, it rotates them by 90 degrees, but they are aligned correctly

MariMakes, 2023-01-11:
Hey @evilC,

I brought your issue up to the team. Your extra notes were super helpful 🙏 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10149

Thanks for the report! 👍

closed 2024-03-01 by wawanbreton