Ultimaker/Cura #14240: Cant Slice the Cinderwing #D Nighwing Curled.

karcaw, 2023-01-08:
### Application Version

5.3.0-alpha+xmas

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. loaded STL.
2. Tried to slice it.

### Actual results

Says slicing failed with unexpected error

### Expected results

its slices
[cura.log](https://github.com/Ultimaker/Cura/files/10367544/cura.log)
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

no project as it would contain the dragon stl which i dont have license to share

karcaw, 2023-01-08:
Slices Fine in Cura 5.2.1

karcaw, 2023-01-10:
So..  as luck would have it I loaded the same model up to try moving and scaling it, and I slices just fine today, at the origin, and off the origin.  so I'm not sure what I changed, as its just my standard Print Profile.  Probably can close the ticket until I see the issue again.

MariMakes, 2023-01-10:
Hey @karcaw,

Thanks for getting back to us.
Good luck with printing your amazing dragon 🐉 

I'll close the ticket.

closed 2023-01-10 by MariMakes