Ultimaker/Cura #14241: Repport

SamRlaput, 2023-01-08:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Alfawise u20

### Reproduction steps

i just slice an stl file and this message appears.

### Actual results

i just slice an stl file and this message appears.

### Expected results

i just slice an stl file and this message appears.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-08:
Thanks for the report.
Let me make sure I have all the facts straight.
You just slice an stl and a message appears?

SamRlaput, 2023-01-09:
Hi. Yes just an stl file.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

MariMakes, 2023-01-10:
Hey @SamRlaput,

We need some more information to troubleshoot 😉 

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-01-25 by github-actions[bot]