Ultimaker/Cura #14243: Add object labeling feature to Cura

KamilLenica, 2023-01-08:
### Is your feature request related to a problem?

I'd like to see object labeling feature to be able to cancel printing one of many object sliced to one gcode file.

This is described in mainsail documentation
https://docs.mainsail.xyz/features/exclude_objects

### Describe the solution you'd like

Analogical solution to egsisting in PrusaSlices / SuperSlicer would be perfect

### Describe alternatives you've considered

Alternative would be to switch to another slicer, but i would like to continue using Cura

### Affected users and/or printers

Anyone using Klipper with Mainsail

GregValiant, 2023-01-08:
This is from the link you provided:
_"Cura & Ideamaker label objects by default in the G-Code, and they are automatically compatible without any further settings."_

KamilLenica, 2023-01-09:
Yes I am dumb I've searched for setting in cura, did not find it and assumed it does not have such feature

closed 2023-01-09 by KamilLenica

jwhite, 2023-05-14:
A bug in Cura 5.3.0 broke some of the exclude_object functionality.  Fixed in Cura 5.3.1.

I discovered this when trying adaptive mesh.  Posting here as there is no obvious reference from bug->fix.

Liionboy, 2024-04-05:
and still nothing, i am sorry but i will go with OrcaSlicer who run very nice with Adaptive Mesh, nobody are interested to make a change on Cura

sesse, 2024-08-21:
It would be nice to have it in Cura without any plugins, but in the meantime, [preprocess_cancellation](https://github.com/kageurufu/preprocess_cancellation) will do this for you (use it with [klipper_preprocessor](https://github.com/pedrolamas/klipper-preprocessor)).