Ultimaker/Cura #14244: Cura 5.3 alpha stops during slicing and is unresponsive after cancelling

Triffen, 2023-01-08:
### Application Version

5.3.0 alpha

### Platform

Manjaro Linux, KDE Plasma

### Printer

Ender 3

### Reproduction steps

1. Start slicing the project
2. Slicing stops about 1/3 of the way
3. Click 'Cancel'

### Actual results

Cura is unresponsive.

### Expected results

Slicing produces gcode; after cancelling Cura stays responsive.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10369384/cura.log)
[Sprite_Cable_Holder.zip](https://github.com/Ultimaker/Cura/files/10369387/Sprite_Cable_Holder.zip)

DerGenaue, 2023-01-08:
Could this be the same issue as https://github.com/Ultimaker/Cura/issues/14133?

Triffen, 2023-01-08:
It's not that: support pattern was Lines; Limit Branch Reach was Off, and while Tip Diameter was 0.6mm, setting it to 0.85mm didn't change anything.
Support should be set to Tree and Support Placement to: Touching Buildplate, otherwise it slices normally.

Triffen, 2023-01-09:
There seems to be a problem with settings saving to profiles as Tip Diameter was never changed before switching from 0.6 to 0.85, and never was 1.2. There were no red lines and no 'Unable to slice' errors.

I've just saved the project with these settings:
[image]
Here's the project: 
[Sprite_Cable_Holder_1_085TipTrianglesNoLimit.zip](https://github.com/Ultimaker/Cura/files/10376048/Sprite_Cable_Holder_1_085TipTrianglesNoLimit.zip)
Clicked Slice, it's  predictably looks like this: 
[image]
Clicked Cancel, clicked Slice again (Cura has frozen, terminated it with the system prompt)
And here is the log: 
[cura.log](https://github.com/Ultimaker/Cura/files/10376066/cura.log)

Triffen, 2023-01-09:
I've also imported this same project, and it imported normally: 4 settings changed from profile ones, all of them right, so 1.2mm tip isn't reproduced for me.

Triffen, 2023-01-10:
Apparently I was using a pre-alpha version still, and forgot about it. The current alpha works fine with the same models and with 'Touching buildplate'. Still not sure what was happening with profiles but that isn't a part of the current bug, so I'll close it.

closed 2023-01-10 by Triffen

ThomasRahm, 2023-01-10:
Profiles are saved as which settings were changed compared to the default settings. So if your version has other default settings than mine, the settings you didn't change will have other values for me.

DerGenaue, 2023-01-10:
For reference: this was the bug that caused this: https://github.com/Ultimaker/CuraEngine/commit/3c4180381b8ecf503c299cf65b442a2ccd587266