Ultimaker/Cura #14245: Cura 5.3.0 Alpha Tree Support slicing stuck

MasterJubei, 2023-01-08:
### Application Version

5.3.0 Alpha

### Platform

Windows 11

### Printer

Creality CR-10 Smart Pro (Improted from creality software)

### Reproduction steps

1. Set support to tree
2. Slice

### Actual results

It will hang in the middle of slicing

### Expected results

Should slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Feel free to mark as duplicate if it's a known issue. Let me know if you need any other details.

[CCR10SPRO_Lucario.zip](https://github.com/Ultimaker/Cura/files/10369403/CCR10SPRO_Lucario.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10369412/cura.log)

DerGenaue, 2023-01-08:
Could this be the same issue as #14133?

MasterJubei, 2023-01-09:
> Could this be the same issue as #14133?

Indeed it is! Thanks, sliced without a problem after disabling the setting.

MariMakes, 2023-01-09:
Excellent troubleshooting! 🎉 
Happy to hear you are unstuck. 

We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14133