Ultimaker/Cura #14247: Launching Cura on a second monitor will crash Cura.

ClaytonM223, 2023-01-08:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

1. Open Cura with second monitor attached. (Cura defaulted to opening on second screen.)

### Actual results

A Cura instance will attempt to open but the screen just stays white and says not responding. I found I can get around the problem by unplugging my second screen, opening Cura like normal, then plugging the second screen back in and dragging the window from the main screen to the second screen.  

### Expected results

Cura should open like normal.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10369724/cura.log)
If it would help to know I am using a Alienware M15 R5 laptop connected via a HDMI to DVI converter to a Acer S200HL with a screen resolution of 1600x900.

MariMakes, 2023-01-09:
Hey @ClaytonM223,

Thanks for your report 👍

We've had a number of reports on this issue, we see it happen with some not all multiple monitor set-ups. 
Most of the issues are collected here https://github.com/Ultimaker/Cura/issues/11907

There are also these workarounds that are surfacing. Can you try and look if these work for you? 
	1. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	2. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	3. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

This seems to be a duplicate of  https://github.com/Ultimaker/Cura/issues/11907
We have a ticket on our backlog with the intent to improve this behavior but resolving this is a little tricky since we are dependend on a library in QT.  

I'll be closing this issue as a duplicate but you can follow the progress here:  https://github.com/Ultimaker/Cura/issues/11907

fieldOfView, 2023-01-09:
> Do you have a Ryzen Graphics card?

Hey @MariMakes, I know this is off-topic here, but there is no such thing as a Ryzen Graphics card. Ryzen is a CPU architecture, not a GPU/graphics card. That whole support article makes no sense because it confuses CPU and GPU. As far as I know, the issue is with the (drivers or firmware for) some Ryzen CPU motherboards. I have mentioned this before, but perhaps you could push the right people to correct the article.

ClaytonM223, 2023-01-09:
This is not the first time I have heard of the sound driver causing problems. I expect that to be a part of my issue. I don't have an AMD graphics card or a GPU overlay on.

fieldOfView, 2023-01-10:
> He means Radeon, not ryzen. simple typo.

No, _she_ means the Ryzen CPU, not the Radeon GPU. The issue that this refers to is this: #5265. That particular issue has nothing to do with the graphics card. It is a matter of someone seeing "AMD" and thinking "graphics card", and other people copying it from there. I am gently reminding @MariMakes that restating that a Ryzen graphics card is somehow involved makes Ultimaker look ... silly? And it is not helpful for users potentially being affected by that issue.