Ultimaker/Cura #14249: Elegoo Neptune 3 Pro not able to Pause at Height or for Filament Change

BenLBurke, 2023-01-09:
### Application Version

4.8

### Platform

Windows 11

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. Sliced my model to find the layer I wanted to pause at
2. Added a "Pause at Height" step in post processing in my GCode and set a specific layer; I also I have tried the filament change option as well.

### Actual results

3. Sliced the file, viewed the GCode in a text editor and didn't see the code to pause
4. Put my model in my printer, the printer does not pause.

### Expected results

I would expect the printer to pause and move to the X/Y coordinates and wait for me to change the filament.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Elegoo_Metric_screw_remix_stop.zip](https://github.com/Ultimaker/Cura/files/10370288/Elegoo_Metric_screw_remix_stop.zip)

GregValiant, 2023-01-09:
Thanks for the report.
Cura 4.8 is several versions past and there are no patches, just new releases.  I would suggest updating to 5.2.1.
I use Pause At Height a lot and as you noted there is no "PauseAtHeight" in that gcode.  There is also no line "Post-Processed" which the PauseAtHeight plugin adds to a file.  So I'm guessing something was amiss when you attempted to add the pause to the file because the file was not post-processed.  That can happen if you mistakenly entered a layer number that is above the top layer of the print.
I can only reproduce this if I enter a layer that doesn't exist.  Can you confirm that you entered a layer that should have been acted upon?

BenLBurke, 2023-01-09:
Thanks Greg. my current model is 50 layers and entered a pause at layer 48.

What i'm realizing is that I have an Elegoo version of Cura, it seems the
Neptune 3 Pro is only supported in their version so I wonder if this is
less of a Cura and more of an Elegoo problem.

Don't know if there is something else I should be doing, downloading an
additional plug in, etc.

I do have a ticket sent to elegoo support as well.
Ben Burke
716.467.2739

GregValiant, 2023-01-09:
UM Cura doesn't actually support resin printers.  I suppose there are a few folks who have gotten it to work.  The proper recourse would be to go back to Elegoo since it is their fork of Cura with the issue.
This same sort of issue comes up with the Creality Slicer and a couple of others that folks have forked off from UM Cura.
I'll go ahead and remove the bug label and close this.  If you have a problem with that let me know and I can re-open this.

closed 2023-01-09 by GregValiant

DD6VD, 2023-03-15:
I have write to the support, and that the answer from 15.03.2023. Take a look :

Hi dear customer,

Unfortunately, currently neptune 3 pro does not support M0/M25 (Pause gcode), the latest version of firmware supports M600 (change filament), you can get the latest version from the link below. For the steps to update the firmware, please refer to the attached document

https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max

Kind Regards
Johnson
Technical Support