Ultimaker/Cura #14250: Cura 5.2.1 Crashes in Preview

AlphaSith, 2023-01-09:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. I upload stl file
2. I make configurations of printing
3. I click on preview button above to see the structure of the thing that I will print
4. I start traveling between layers using the bar on the right side of the screen

### Actual results

Cura freezes while traveling between layers for a few seconds then my screen goes black for 1 second and when it comes back cura window is all white for a few seconds and then it crashes and I'm back at my desktop screen. Doesn't matter which file I try to slice it happens with every stl files.

### Expected results

The expected result from cura is not to crash and just let me see the layers.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10371911/cura.log)

MariMakes, 2023-01-09:
Hey @AlphaSith,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I've only heard this happen with really large and intricate models but they never crash, often they are just slow. 
Have you been able to see the preview of this model in other versions of Cura before? 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

If that's not possible can we at least get a screenshot or a screen recording? 
Thank you 🙏

AlphaSith, 2023-01-09:
[curaprojectfile.zip](https://github.com/Ultimaker/Cura/files/10377423/curaprojectfile.zip)

https://user-images.githubusercontent.com/122265248/211422458-dab4670b-d4ef-45fd-a522-6bc9ae90ce37.mp4

Hello,
Here's a project file and a video of issue. I didn't have any problems with Cura 4.5  but after updating to cura 5.2.1 this thing started happening.  As I said STL files or setting doesn't matter when I go to preview and start inspecting layers it crashes.

AlphaSith, 2023-01-11:
Hey @MariMakes,
You have any ideas?

MariMakes, 2023-01-11:
Hey @AlphaSith,

Sorry I didn't get back to you. 
I noticed that the project file is not loading your model for me and lost track of the ticket. 
I just have an empty buildplate, but the correct printer and material. 

Can you double-check if you included the correct projectfile?
Perhaps try to include it again?

AlphaSith, 2023-01-12:
[CE3PRO_Body.zip](https://github.com/Ultimaker/Cura/files/10402070/CE3PRO_Body.zip)
@MariMakes Here's one with a model on buildplate if you need.

MariMakes, 2023-01-12:
Hey @AlphaSith,

I took another look and I can't reproduce your issue on my side. I'm not getting any crashes with your model or settings. 
At first I thought it was related to a missing bundled plug-in but that doesn't seem to be the case. 

I'll have to bring it up tot the team to take a look, but that might take a while. 

To help you get unstuck...
1. Can you check if compatibility mode is working for you? You can find it under Preferences | Preferences| Force Layer View Compatibility Mode (restart Required)
2. Can you check if your OpenGL driver is up to date?

AlphaSith, 2023-01-12:
Hi @MariMakes,
I have checked them both.
1) OpenGL driver is up to date.
2) Force Layer View Compatibility Mode was unmarked (off), so I opened it and restarted Cura. Now Cura just freezes when I travel between layers. Same way as before but now it just freezes instead of crashing.

AlphaSith, 2023-01-15:
Hey @MariMakes,
Any news?

casperlamboo, 2023-01-17:
Hi @AlphaSith,

Thank for bringing this issue to our attention. I read through the logs but unfortunately did not find any issues. This is somewhat to be expected as cura just closes (normally a crash window appears). There was probably not a change to write the exact error to the logs.

One thing you might do is remove the cura cache folder located at `C:\Users\[USER]\AppData\Local\cura\5.2\cache`.

Previously we had some issues regarding amd GPU's. What CPU/GPU combination are you using?

When reading your comments it looks like this behavior is really consistent, each time you slice and look through the model in layer view the application crashes. Is this assumption correct or is it crashing more sporadically?

casperlamboo, 2023-01-19:
Hi @AlphaSith,

Since we cannot find any indication of the crash in the logs and we have no way of reproducing the problem (the error does not occur on any machine we have available) we are unfortunately not able fix the issue at this moment. We will keep an eye open for similar reports and when they come in we can hopefully narrow down the cause of the issue.

Hope you understand our decision.

Thank you,
@casperlamboo

closed 2023-01-19 by casperlamboo

Mannicris, 2023-05-15:
Hi @AlphaSith  & @casperlamboo ,

I have the same exact issue and also the same CPU/GPU combination Core i7-11800H, Nvidia RTX3070 Laptop GPU.
Did any one of you find a solution for this problem?

It is really difficult to work with cura like this, because the crash happens almost every time.

tarhunakau, 2023-09-23:
Why do it closed? The issue is still actual. I can clice only basic models but if a model has some details - it crashes. The only way to slice - use my second laptop with MacOS.

Windows 11, i5-11400, RTX3600.