Ultimaker/Cura #14252: Cura Slicing Bug

moneymason0, 2023-01-09:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Trying to slice for multiple parts. And it failed slicing

### Actual results

[Cura Slicing Bug.zip](https://github.com/Ultimaker/Cura/files/10375223/Cura.Slicing.Bug.zip)

### Expected results

Expected the program to slice

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

N/A

GregValiant, 2023-01-09:
Thanks for the report and the file.
The file wouldn't slice for me either.  I selected all the models and "grouped" them and rotated the group 90° on the cura build plate.  That allowed it to slice.
This is a duplicate of numerous other reports where good models won't slice unless rotated or moved on the build plate.
I'll mark this as a duplicate slicing bug and go ahead and close it.  The Cura team has a lot of examples of this behavior and hopefully it will be fixed in 5.3.

closed 2023-01-09 by GregValiant

moneymason0, 2023-01-09:
Greg,

Thanks for taking the time to look at this. I did your suggested fix and it
worked!

Saved me from going insane, life saver!

Very Respectfully,

Mason