Ultimaker/Cura #14253: Support overhang angle

LittleBeast11, 2023-01-09:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Loaded the STL file now the areas that generally show red (which require supports) is showing on the top of the model
[image]

### Actual results

The prepare screen shows the red faces on the top of the model and not where the supports are required

### Expected results

The red should be on the other side f the model as it should show which areas require supports

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I have tried multiple files and they all do the same

GregValiant, 2023-01-09:
When surfaces show red on the top of the model - that indicates flipped surfaces in the model itself.  Those need to be repaired before you can slice the model.  MS 3D Builder should be able to make the repairs.
When Cura encounters a flipped normal then the slicer can no longer figure out the inside from the outside and the slice will fail or just come out wrong.
If you would load the "Mesh Fixes" plugin from the Marketplace it has some tools that you might be able to fix your model with.
So this isn't a bug but rather it is expected behavior to inform the user that there is a problem with the model.
I'll remove the bug label but I'll leave this open so you can respond here.

LittleBeast11, 2023-01-10:
Hi Greg,

Many thanks for the prompt response.

Will give it a try.

Warm regards

*Vincent Kruger*
*084 811 7117*

GregValiant, 2023-01-10:
I think if you view that model in "X-Ray" view then the red problem areas would be highlighted better.
I'm going to close this as it does appear to be a problem with the model.

closed 2023-01-10 by GregValiant