Ultimaker/Cura #14257: Created a usefull new infill pattern using unconventional settings

printingotb, 2023-01-10:
### Is your feature request related to a problem?

I found a combination of infill settings that creates a supporting wall or "tube" for skin edges.
Settings used:
*Concentric infill at 0.01% (effectively disables infill while still allowing extra infill walls and access to infill settings)
*Extra infill wall count = 1
*Infill support 89% (90 wont work)

See pictures. I believe it can save a lot of material while supporting only what's needed. More stable and less travels than lightning infill. Possibly using even less material. More infill walls can be used to make it very strong and able to handle load.
This can be used together with 1-2 extra skin support layers to create a very good bond with the skin lines.
It could also be useful to have these supporting tubes present within conventional infill.

### Describe the solution you'd like

Generate these tubes separately from other infill "Infill tubes" or something better named.
The key to make them useful is to make them only affect horizontal surfaces (infill support 89% in this case) but we may still want  regular infill at the same time.

### Describe alternatives you've considered

In the past, I have tried creating a similar result by setting Skin support layers to 9999 but it often makes Cura crash (and since the setting isn't intended to be used that way, I haven't reported it as a bug). This also creates a ton of skin support under sloped surfaces.

I imagine this would be difficult to implement in a good way and you have probably had the same ideas already but I thought I would share it :smiley:

### Affected users and/or printers

Any

### Additional information & file uploads

[image]
[image]
[image]
(Portions of the model removed)
[paw cut.3mf.zip](https://github.com/Ultimaker/Cura/files/10383394/paw.cut.3mf.zip)

5axes, 2023-01-19:
Interesting ...  But How do you deal with the support of the skin layer and his border ?
[image]

Do you expect something like that ?

[image]

printingotb, 2023-01-19:
I actually don't want any infill there as it prints well without it. But if we could get the infill tubes in my example at the same time as normal infill (without adding more infill walls to the tubes), we could support those areas too. The tubes will add to the vertical load capacity, reducing the infill % needed for a strong part. It will also reduce the number of top layers needed to close an area since the first top layer will not fall down around the holes.