Ultimaker/Cura #14258: Cura 5.2.1 modern linux crashes when trying to add Ender 3 S-1 profile

oliof, 2023-01-10:
### Application Version

5.2.1, 5.3.0-alpha

### Platform

Linux pop!_os (ubuntu 22.04 derivate)

### Printer

Ender 3 S-1

### Reproduction steps

1. Went to Settings -> Printer -> Add Printer
2.  select "Add a non-networked printer"
3. Choose "Creality Ender-3 S1" from section without title
4. click "Add"

[image]

### Actual results

Cura Crashes: When starting the AppImage from the command line, I get a "Segmentation fault (core dumped)" error.

### Expected results

Cura should add the printer like others

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Pertinent lines from cura.log:

``` 
2023-01-10 16:50:30,225 - WARNING - [MainThread] UM.Qt.Bindings.Theme.getIcon [154]: The icon 'category_machine' is deprecated. Please use icon 'Printer' instead.
2023-01-10 16:50:30,244 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2023-01-10 16:50:30,642 - DEBUG - [JobQueueWorker [1]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2023-01-10 16:50:30,730 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [221]: Error check finished, result = False, time = 3.75s
2023-01-10 16:50:35,659 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_CurafZDMX9/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:53:9: QML ViewMenu: Binding loop detected for property "width"
2023-01-10 16:50:35,659 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_CurafZDMX9/share/cura/resources/qml/MainWindow/ApplicationMenu.qml:53:9: QML ViewMenu: Binding loop detected for property "width"
2023-01-10 16:50:39,333 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._check_unmapped_settings [79]: called
2023-01-10 16:50:39,334 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFPlugin._check_unmapped_settings [96]: No unmapped settings found.
2023-01-10 16:50:39,676 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_CurafZDMX9/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:237:25: Unable to assign QString to QRegularExpression
2023-01-10 16:50:40,354 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.018189430236816406 seconds
2023-01-10 16:51:00,363 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_CurafZDMX9/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:218:21: Unable to assign [undefined] to QString
2023-01-10 16:51:00,364 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_CurafZDMX9/share/cura/resources/qml/WelcomePages/AddLocalPrinterScrollView.qml:207:21: Unable to assign [undefined] to QString
2023-01-10 16:52:36,464 - INFO - [MainThread] cura.Settings.MachineManager.addMachine [425]: Trying to add a machine with the definition id [creality_ender3s1]
2023-01-10 16:52:36,596 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [308]: The definition cache for definition creality_ender3s1 failed to pickle.
2023-01-10 16:52:36,749 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3s1 took 0.14974427223205566 seconds.
2023-01-10 16:52:36,753 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createExtruderStackWithDefaultSetup [108]: Exception: Unable to find extruder definition with the id [creality_base_extruder_0]
2023-01-10 16:52:36,753 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createExtruderStackWithDefaultSetup [108]: Traceback (most recent call last):
2023-01-10 16:52:36,753 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createExtruderStackWithDefaultSetup [108]:   File "cura/Settings/CuraStackBuilder.py", line 104, in createExtruderStackWithDefaultSetup
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createExtruderStackWithDefaultSetup [108]:     extruder_definition = registry.findDefinitionContainers(id = extruder_definition_id)[0]
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createExtruderStackWithDefaultSetup [108]: IndexError: list index out of range
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: Exception: Failed to create an extruder stack for position 0: Unable to find extruder definition with the id [creality_base_extruder_0]
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: Traceback (most recent call last):
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]:   File "cura/Settings/CuraStackBuilder.py", line 104, in createExtruderStackWithDefaultSetup
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]:     extruder_definition = registry.findDefinitionContainers(id = extruder_definition_id)[0]
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: IndexError: list index out of range
2023-01-10 16:52:36,754 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: 
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: During handling of the above exception, another exception occurred:
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: 
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: Traceback (most recent call last):
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]:   File "cura/Settings/CuraStackBuilder.py", line 67, in createMachine
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]:     cls.createExtruderStackWithDefaultSetup(new_global_stack, position)
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]:   File "cura/Settings/CuraStackBuilder.py", line 109, in createExtruderStackWithDefaultSetup
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]:     raise IndexError(msg)
2023-01-10 16:52:36,755 - ERROR - [MainThread] cura.Settings.CuraStackBuilder.createMachine [69]: IndexError: Unable to find extruder definition with the id [creality_base_extruder_0]
2023-01-10 16:52:36,756 - WARNING - [MainThread] cura.Settings.MachineManager.addMachine [438]: Failed creating a new machine!
```

fieldOfView, 2023-01-10:
We're talking semantics here, but it is due to _your use of_ Startup Optimiser. After installing that plugin, you told it to remove unused profiles. It stands to reason that you will no longer have a full set of profiles after that.

The intended workflow of the plugin if you want to add a printer you have not added before is to use the menu item in Extensions -> Startup Optimiser to restore all profiles, then restart and add the new printer. After that you can disable unused profiles again.

I'm closing this because it is not really an issue with Cura (and the plugin that caused the issue is functioning as intended here).

closed 2023-01-10 by fieldOfView

oliof, 2023-01-10:
You are absolutely right of course, I should have closed the issue myself because PEBKAC (-: