Ultimaker/Cura #14259: ERROR: Cannot wrap command with different envs

wasmann, 2023-01-10:
### Application Version

5.1.2

### Platform

windows 10

### Printer

no printer needed

### Reproduction steps

I tried to build the fork from soure using anaconda.
When I excuted the command line " conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv", I got "ERROR: Cannot wrap command with different envs" error.
[image]

### Actual results

ERROR: Cannot wrap command with different envs, ['C:\\Users\\waz0\\workspace\\Cura\\venv\\conan\\conanrun.bat'] - [] - ['C:\\Users\\waz0\\workspace\\Cura\\venv\\conan\\conanrun.ps1']

### Expected results

Can you help me build from source code hand by hand please. Because I am not so expert on that.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I cant run it from source using anaconda as well. It gave an error saying UM not found but I have added it in the environment path already ( successfully make it work on powershell).

wasmann, 2023-01-20:
hello

MariMakes, 2023-05-04:
Hey @wasmann,

Sorry, it took us a while to get back to you 😞 
😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

This is a neat trick to keep our Github issues more manageable. 

I'm super curious. What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

closed 2023-05-30 by github-actions[bot]

pietchaki, 2023-08-25:
I was having the same issue, following the previously mentioned steps to get a clean slate didn't help.

Solution was to delete the "venv" folder and all worked fine as this folder is created by the _conan install_ command

rburema, 2024-09-10:
For future reference (or however long we'll still be on Conan 1.x at least):

So this appears to be (or is treated like) a bug fixed in Conan 2.x, as far as I can see at least (see [here](https://github.com/conan-io/conan/pull/13084) which I found because someone posted a similar problem w.r.t. building boost, [here](https://github.com/conan-io/conan/issues/12764))  -- we'll be on Conan 2.x soon-ish, but in the mean-time:

As a quick-fix, you can also just delete either the `.bat` or `.ps1` file it refers to, depending on whether you're running in cmd-prompt (or 'Native Tools Command Prompt' when using VS) or PowerShell.