Ultimaker/Cura #14261: Unable to open Gcode files

GregValiant, 2023-01-10:
### Application Version

5.2.1 and 5.3

### Platform

Windows 11

### Printer

Any

### Reproduction steps

Attempt to open a gcode file.  Received an error message.
[image]

Went to the MarketPlace and disabled the Gcode Reader plugin.  Retsarted Cura.
Re-enabled the Gcode Reader plugin.  Restarted again.
Still receive the error for all gcode files.

4.13 loads the gcodes as expected.

### Actual results

Error as noted above.
This worked until a couple of days ago.  The first time it happened - disabling and re-enabling the plugin brought it back.  It was OK for a couple of days then last night there was an update to Windows and the Gcode Reader no longer functions.  It still works in 4.13.1.

### Expected results

Open the gcode file for viewing.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10385258/cura.log)

This is the first error line from the attached log
2023-01-10 12:19:27,404 - ERROR - [JobQueueWorker [3]] UM.FileHandler.ReadFileJob.run [80]: Exception: Exception occurred while loading file C:/Users/Greg/Documents/Creality/gcode/8colors.gcode
This is a new computer and it is somewhat confused about which account is me.  That may have a bearing as the Cura.log file is in the "owners" account where I am signed in under my own name.  I have also noticed that the "Help | Show Configuration Folder" command in both 5.2.1 and in 4.13.1 do not work.  No errors, just no response at all.

Let me know if you require more information.

GregValiant, 2023-01-11:
The problem occurred when attempting to open a multi-extruder gcode with a single extruder printer active in Cura.  The file would not open and seems to have caused a glitch in the G-Code reader.  Any subsequent gcode file also refused to open.  Shutting Cura down and starting it back up seems to fix the problem.  

With the multi-extruder printer active then all gcode files opened as they should.
With the single-extruder printer active then opening a multi-extruder gcode file causes a crash in G-code Reader.

MariMakes, 2023-01-16:
Hey @GregValiant,

Thanks for your report. 😮  I wasn't aware that it was broken. 
I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

MariMakes, 2023-01-19:
Hey @GregValiant,

Thanks for bringing this to our attention 🙏 

We've added a ticket to the backlog with the intent to improve this hopefully soon! 😨 
For internal reference CURA-10183

Thanks for the report! 👍

GregValiant, 2023-11-14:
@casperlamboo Thanks.  This appears to be fixed so I'll go ahead and close this ticket.

closed 2023-11-14 by GregValiant