Ultimaker/Cura #14262: Generated with Cura_SteamEngine

David-4l, 2023-01-10:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

ELEGOO NEPTUNE 2

### Reproduction steps

In the Gcode file, there is the start Gcode, the end Gcode and a part Generated with Cura_SteamEngine.
The problem is that Cura generates a command that did not exist in version 4.8 it adds the command G1 F600 Z0.45 which creates disorder as soon as printing starts.

### Actual results

The nozzle wipes the filament and rises to go to print, with this command in excess, the nozzle goes down in its filament deposit.

### Expected results

The nozzle wipes the filament and rises to go to print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-10:
Thanks for the report.
Do you have Z-Hops enabled?  Maybe at 0.25 height?

David-4l, 2023-01-10:
that's right, at 0.25mm yes.
is there a link ?
Sorry im french

GregValiant, 2023-01-10:
Yes.  That is the Z-hop in response to the retraction about 6 lines above.  In your case the Z was at 2.0 and so it is a move downward instead of up.  There is no "E" involved.
The next line is the travel to the start of the first extrusion of the skirt.
Then there is the "Z-hop down"
Then the first extrusion starts.
This is all expected behavior if Z-hops are enabled.  They shouldn't cause any trouble as the move would be made anyway but as a straight line to "G0 F7500 X104.05 Y101.729 **Z0.2**".  Functionally there is no difference as the nozzle ends up at the same place.  In your case the travel is at .45 instead of angling from Z2.0 to Z0.2.  The extruder is not involved either way.

David-4l, 2023-01-10:
I just disabled Z-Hops, the G1 F600 Z0.45 command line disappeared.
What is the connection, why 0.45 and how to have Z-Hops without the Z0.45 command line
Thanks for your help

GregValiant, 2023-01-10:
The setting is "Z-Hop When Retracted".  Whenever there is a retraction there will be a Z-hop.
Your "Initial Layer Height" is 0.20.  The Z-hop height is 0.25.  0.2 + 0.25 = 0.45 and so that is the height that the travel move occurred.
A perfect start to the skirt is difficult to attain.  There is going to be some lag as the print gets going.  That z move to .45 isn't the problem.  Whether it travels across at .45 or at .2 the move is going to be made and is going to take just as long either way.

David-4l, 2023-01-10:
It didn't seem to me that Initial Layer Height was 0.2mm.
What bothers me is that the nozzle will move towards the center of the part at 0.45mm from the plate (not great), I wanted the nozzle to be at 2mm and gradually go towards the center to finish at 0.2mm before launching his cycle.

I understand that the starts are never easy to program

GregValiant, 2023-01-10:
;TYPE:SKIRT
G1 F600 Z0.2
So the initial layer height was 0.20.
The function of a skirt is to get the printer properly primed.  After that it has no function and is scrapped.  So long as during it's course it sticks to the build plate and any extrusion problems end up in the skirt instead of in the print, things are good.
If you are going to use Z-hops then that "G1 F600 Z0.45" line will be in the gcode.  You can make a couple of manual changes.
Comment out the Z-Hop by putting a semi-colon in front.  Change the following line to Z0.2.

;G1 F600 Z0.45
G0 F7500 X104.05 Y101.729 **Z0.2**

At any rate what we have here is expected behavior given that you have Z-hops enabled.  I'll remove the bug label and leave this open so you can respond.