Ultimaker/Cura #14263: Prime tower when only one extruder in use

therealzoomgod, 2023-01-10:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom IDEX

### Reproduction steps

Set up an IDEX print with a prime tower, one extruder for model and second extruder handles support.
Change the print so only 1 extruder is used for model and support

### Actual results

It's no longer an IDEX print but the prime tower remains and prints with whatever extruders it was enabled on.

### Expected results

I would think a non IDEX print should disable the IDEX options to simplifiy things.  I ran across this while writing a post processing script specfic to IDEX prints for klipper macros.  Makes scripting difficult as it shows 2 extruders in use when there should be one.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[3 scenarios in .3mf files.zip](https://github.com/Ultimaker/Cura/files/10386853/3.scenarios.in.3mf.files.zip)

GregValiant, 2023-01-11:
Thanks for the report.
The Prime Tower Brim is treated as a part of the overall Build Plate Adhesion and not part of the prime tower.  So although you aren't using the prime tower, the brim will print.  You can turn off the PT brim but leave the prime tower enabled and you will see that in the first two files, the tower brim disappears.  That is expected behavior and has annoyed some folks before.  If you leave the PT enabled and turn off the PT brim, and use a skirt, you will see that the skirt goes around the PT location even though nothing gets printed there.  I think it's the kind of decision you don't want the software making for you as every print is going to be a least a little different.  In regards to the first file that doesn't use E2; even if you had disabled E2 then there wouldn't be a prime tower, but there would be a PT brim.
I'll leave this as a bug report so the Cura team can respond.

therealzoomgod, 2023-01-22:
If I recall there is no comment in gcode for prime tower/brim when only one extruder is used.  That seems to me at least that the idea was for it to be turned off.