Ultimaker/Cura #14264: Brim issue with large/tall objects (>50 cm)

ThijsBeckers, 2023-01-10:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Tronxy Veho-600

### Reproduction steps

Import a tall model (55 cm).
Add a brim at 0cm.
Slice.

### Actual results

A very small brim is generated that has nothing to do with the footprint of the model.

### Expected results

A 'normal' brim should be created.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Faulty Brim.zip](https://github.com/Ultimaker/Cura/files/10386996/Faulty.Brim.zip)
[image]

GregValiant, 2023-01-10:
Thanks for the report.
The bottom of the model isn't flat.  There is a piece of thread sticking down past the bottom .52mm and that's the point Cura is setting as Z=0 on the model.  So you are only getting the brim around that tiny point.
[image]

If you set the model to Z=-0.5 in Cura you should be good to go.  You could also alter the model to make the bottom flat.
I'll remove the bug label and close this as it's a model problem.  I hop you understand.

closed 2023-01-10 by GregValiant

ThijsBeckers, 2023-01-12:
Hi @GregValiant,

Thank you A LOT! I honestly didn't notice 😖. I REALLY appreciate you for answering this so quickly! I'm going to correct the model and I'm sure it'll work. And I apologize for taking your time while this was obviously my own dumb mistake.

Cheers and I hope you keep up the good work!

GregValiant, 2023-01-12:
I knew what the problem was as soon as I saw the post.  It took some fooling around to find the little bugger though.  It's just a tick but computers are precise.