Ultimaker/Cura #14265: Slicing failed on STL from thingiverse

ronguest, 2023-01-10:
### Application Version

5.2.1

### Platform

macOS Ventura

### Printer

Sovol SV01

### Reproduction steps

1. I downloaded the files from [spiderman headphone statnd](https://www.thingiverse.com/thing:4753064/files)
2. I opened "MM_base_UPDATED" and hit slice
3. It failed. The failure is repeatable.

### Actual results

Error popup saying slice failed

### Expected results

A G code file
[MM_base_UPDATED.3mf.zip](https://github.com/Ultimaker/Cura/files/10387295/MM_base_UPDATED.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/10387298/cura.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Seems easy to reproduce wi these files

GregValiant, 2023-01-10:
Thanks for the report.
Just because a model comes from Thingiverse doesn't mean it's a good model.  If you load the Mesh Tools plugin from the MarketPlace you would have received the "Model is not watertight and may not print correctly" warning.
There is also a bug in Cura that can keep a model from slicing when the model is at the build plate midpoint.  I moved the model and rotated it and it sliced.  I'll mark this as a duplicate and a slicing error.  
[image]

@MariMakes I struggled with this one.  After repairing the model it sliced, and then it wouldn't slice, and then it sliced.  It may be one you want to look at.

ronguest, 2023-01-11:
Thanks for the super fast response! I’ll load that plugin and know what to do if I see something like this in the future.Cheers,RonSent from my iPad

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker