Ultimaker/Cura #14266: Problema no Segmenta

sergiocassini, 2023-01-10:
### Application Version

5.1.0

### Platform

windows 10

### Printer

Hellbot Magna 2 230

### Reproduction steps

Puse a cortar una figura con dos estrusores en cura para una impresora hellbot Magna 2 230, la misma consta de cuatro capas a diferente altura, al segmentarla se cuelga.

### Actual results

[image]

### Expected results

Se termine de segmentar

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-01-10:
Thanks for the report.
We need a project file to troubleshoot what is going on.  Use the "File | Save Project" command to create a 3mf project file.  Zip the file and post the zip folder here.

MariMakes, 2023-01-16:
Hey @sergiocassini,

Lo siento, no puedo hablar español, pero creo que puedo ayudarte.

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-01-31 by github-actions[bot]