Ultimaker/Cura #14268: Cura locks up when run on Monitor 2 of a system with 2 monitors.

theojt, 2023-01-11:
### Application Version

5.2.1

### Platform

Windows 11, Ryzen 9 5900, 64GB DDR4, Nvidia GeForce RTX 2060, 2X ASUS 1920x1080 monitors

### Printer

Ankermake M5

### Reproduction steps

1.  I Attempted to Open Cura on Monitor 2 - but it just hangs and must be terminated/restarted.
2.  If I disable Monitor 2 (forcing Cura to use Monitor 1, Cura opens successfully on Monitor 1.
3. If I subsequently reenable Monitor 2, Cura continues to work on Monitor 1, unless I do something that opens a new window.  
The new window then opens on Monitor 2 and the app freezes once again.

### Actual results

As noted above, Cura fails when run on Monitor 2

### Expected results

Cura should run without issues on either Monitor 1 or Monitor 2.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have received my Ankermake M5 and it seems that a modified version of Cura is used by Anker for its Slicer.  This problem also exists within that software (Ankermake v0.9.2_3) though it behaves slightly differently.

MariMakes, 2023-01-16:
Hey @theojt,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We've had a number of reports like this. 
Most of them are collected here: https://github.com/Ultimaker/Cura/issues/11907

There have already been some workarounds
1. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
2. You seem to have a Ryzen. Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
3.. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

theojt, 2023-01-16:
Thank you! **Disabling the Nahimic service immediately resolves the issue.**  Enabling the Nahimic service recreates the issue.
Other workarounds had no impact on my system.

theojt, 2023-01-16:
Hi,

Thank you - my issue is resolved by the work-around.

I tried all the suggested work-arounds:

  1.  Disable GPU overlay - no effect
  2.  Disable Nvidia Overlay setting for the Cura app - no effect
  3.  Disable Nahimic audio service - immediate resolution

I reverted all tested setting back to their original settings and Cura continues working on the 2nd monitor.  Note that re-enabling the Nahimic service re-introduces the issue, disabling it resolves the issue (repeatedly).

Included below are the details of my system.

System:
Mainboard: ASUS ROG Crosshair VIII Dark Hero
CPU: AMD Ryzen 9 5900X
Mem: 64GB DDR4
GPU: Nvidia GeForce RTX 2060

Versions:
Windows 11:  Build 25276.rs_prerelease.230106-1334
Cura: 5.2.1

Display-related:
NVIDIA GeForce Experience: 3.26.0.160
GeForce Game Ready Driver:  528.02
PhysX System Software: 9.21.0713
HD Audio Driver: 1.3.39.16
USBC Driver: 1.50.831.832

3.. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? #12102 (comment)<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FUltimaker%2FCura%2Fissues%2F12102%23issuecomment-1157907665&data=05%7C01%7C%7C426b2711ece64d936f1f08daf7db2378%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638094817811474389%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=oE43h4PqAfMqd43Y7P9ehbwz5tc9p9NjQa7cvH3ZHRE%3D&reserved=0>

MariMakes, 2023-01-17:
Hurray! I'm glad to hear your issue is resolved. 
Thanks for getting back to us. 👍 

I'll be closing the ticket, happy printing!

closed 2023-01-17 by MariMakes