Ultimaker/Cura #14273: [5.2.1] MacOS - saving to a removable disk that is full appears to work

keylevel, 2023-01-11:
### Application Version

5.2.1

### Platform

MacOS Ventura 13.1

### Printer

Ender 5 Pro

### Reproduction steps

Save a sliced model to external media that is either full or which has insufficient space for the gcode file.

### Actual results

Saving the model appears to work, and the eject button is presented. However, this happens even when there is not enough space on the card for the file - in which case two outcomes are possible:

1. If there is insufficient space, the file is truncated (with the result that the print stops part way through).
2. If there is no space, a zero length file is created (which appears to load on the printer, but nothing happens).

### Expected results

Any filename that was added to the external media should be removed (so that an incomplete / empty file cannot be loaded into the printer) and the user should be informed that the card is full.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

MariMakes, 2023-01-16:
Hey @keylevel,

Welcome to the Ultimaker Cura Github 🚀 

Wow! I don't think that this is something we tested.
I'm sorry you found this out the hard way. 😖 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

MariMakes, 2023-01-19:
Hey @keylevel,

Thanks again for the detailed report.  💪 
I brought your issue up to the team. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10180

Thanks for the report! 👍

keylevel, 2023-01-20:
Thanks for the feedback @MariMakes.

MariMakes, 2023-10-11:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪